jimmyyhwu/pose-interpreter-networks #5: blender in pose estimation

dingshenglan, 2019-03-07:
Hello, when I run the demo, visualization, and evaluation code in "pose estimation", the program has been rendering in the background of the blender, unable to proceed to the next step, and there has been no result. I encountered this problem for the first time and put the “mesh” folder in the Oil Change dataset. The permissions of the .stl file in the file are changed to executable files, which can run normally. 
However, after a few days, I met the same problem that the program has been running can not produce results, I don't know how to solve this time.

jimmyyhwu, 2019-03-07:
Could you provide the terminal output from Blender? It should be in the same place you opened `jupyter notebook`.

dingshenglan, 2019-03-12:
> 你能提供Blender的终端输出吗？它应该在你打开的同一个地方`jupyter notebook`。

Hello, I changed the permissions of all .stl files to a+x  after that, and I can run it out. Thanks

closed 2019-03-12 by dingshenglan