jimmyyhwu/pose-interpreter-networks #8: Issue about  training pose estimation models.

yikakabu, 2019-03-30:
Hi, Jimmy.
When I ran 

> python train.py config/floating_kinect1_mask.yml

There was a error

> File "/home/huo/virtuallist/py36/lib/python3.6/site-packages/pypcd/pypcd.py", line 282, in point_cloud_from_fileobj
> if ln.startswith('DATA'):
> TypeError: startswith first arg must be bytes or a tuple of bytes, not str 

came out.

It seems because of loading pcd file incorrectly. I have tried to use the two fomat of pcd file in binary and ascii, but the same error still occurs. And if I modified the code `ln.startswith('DATA')` with `ln.startswith(b'DATA')`, there will be a error 

> File "/home/huo/virtuallist/py36/lib/python3.6/re.py", line 172, in match
>     return _compile(pattern, flags).match(string)
> TypeError: cannot use a string pattern on a bytes-like object

come out subsequently.

Could you please give some adcices to solve the peoblems above?
Thank you.

jimmyyhwu, 2019-03-30:
How did you install `pypcd`? Did you apply the patch to enable Python 3 compatibility?

yikakabu, 2019-03-31:
@jimmyyhwu 
Thank you for your reply. 
I followed the README to install `pypcd` and I did applyed the patch to enable Python 3 compatibility. But my python site-packages also contains a `pypcd` and the train.py calls it by default. So I solved this by uninstalling the `pypcd` in site-packages.

closed 2019-03-31 by yikakabu