jimmyyhwu/pose-interpreter-networks #9: Where to download provided val?

dingshenglan, 2019-04-07:
Hello, in poseprocess_wraper.py:
“parser.set_defaults(process_val=False)  # should download provided val set to match numbers in paper”
Where to download val set??
I set process_val=True to create val dataset ， but when I train my pose estimator：
[image]
I use kinect2.  I just foget to change "experiment_name"

jimmyyhwu, 2019-04-07:
See [here](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/pose_estimation#dataset) for the val set used in the paper. However, if you are training using your own data, you can just turn the `--process_val` flag on and generate your own val set.

dingshenglan, 2019-04-07:
> 请参阅[此处](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/pose_estimation#dataset)了解。本文中使用的VAL集。但是，如果您正在使用自己的数据进行培训，可以则打开`--process_val`标志并生成自己的VAL集。

OK，thanks very much.

closed 2019-04-07 by dingshenglan