jimmyyhwu/pose-interpreter-networks #11: Error about ros package

dingshenglan, 2019-04-10:
Hello, when I run "roslaunch pose_interpreter_networks pose_estimator.launch", I met this error, as follow:

> [ERROR] [1554901200.082931]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fb60d16e910>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 75, in callback
    self.signalMessage(msg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 224, in add
    self.signalMessage(*msgs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/message_filters/__init__.py", line 57, in signalMessage
    cb(*(msg + args))
  File "/home/dsl/catkin_ws/src/pose_interpreter_networks/src/pose_estimator.py", line 117, in callback
    segm, object_names, positions, orientations = self.model(input)
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/modules/module.py", line 477, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/data_parallel.py", line 124, in forward
    return self.gather(outputs, self.output_device)
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/data_parallel.py", line 136, in gather
    return gather(outputs, output_device, dim=self.dim)
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/scatter_gather.py", line 67, in gather
    return gather_map(outputs)
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/scatter_gather.py", line 62, in gather_map
    return type(out)(map(gather_map, zip(*outputs)))
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/scatter_gather.py", line 62, in gather_map
    return type(out)(map(gather_map, zip(*outputs)))
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/scatter_gather.py", line 62, in gather_map
    return type(out)(map(gather_map, zip(*outputs)))
  File "/home/dsl/.conda/envs/pose/lib/python2.7/site-packages/torch/nn/parallel/scatter_gather.py", line 62, in gather_map
    return type(out)(map(gather_map, zip(*outputs)))
TypeError: zip argument #1 must support iteration

How to do to solve this problem?

jimmyyhwu, 2019-04-10:
I have not seen this before. Would it help if you restricted to one GPU using `export CUDA_VISIBLE_DEVICES=0` before running the code?

dingshenglan, 2019-04-10:
Yes！It can work，thank you very much！

closed 2019-04-11 by dingshenglan