jimmyyhwu/pose-interpreter-networks #12: About trainning pose estimator

dingshenglan, 2019-04-13:
Hello, I changed the model in pose estimator from  resnet18 to  resnet50, I use the points loss, but when I train the estimator on th mask dataset, the orientation loss does not converge，could you give me some suggestion？
[image]
[image]
[image]

jimmyyhwu, 2019-04-13:
I have not tried using ResNet-50, you may have to change the hyperparameters for the bigger model. From your plots it looks like the learning rate may be too high.

dingshenglan, 2019-04-13:
> 我没有尝试使用ResNet-50，您可能需要更改较大型号的超参数。从您的情节来看，学习率可能太高了。

Ok，I will use smaller learning rate. Thanks verymuch :)

closed 2019-04-13 by dingshenglan