jimmyyhwu/pose-interpreter-networks #13: Error about "end to end pose estimator" use "kinect2" in Oilchange datasets

dingshenglan, 2019-04-13:
When testing “end to end system” in Oilchange dataset，I used “kinect2“，but  I met  the key error. 

> KeyError                                  Traceback (most recent call last)
<ipython-input-169-fed55fe84c4c> in <module>
      2 orientation_errors = []
      3 with torch.no_grad():
----> 4     for input, target, object_index, object_id in tqdm(val_loader):
      5         position_error, orientation_error = forward_batch(model, input, target, object_index, object_id)
      6         position_errors.extend(position_error)

~/.conda/envs/poseIN/lib/python3.6/site-packages/tqdm/_tqdm_notebook.py in __iter__(self, *args, **kwargs)
    219     def __iter__(self, *args, **kwargs):
    220         try:
--> 221             for obj in super(tqdm_notebook, self).__iter__(*args, **kwargs):
    222                 # return super(tqdm...) will not catch exception
    223                 yield obj

~/.conda/envs/poseIN/lib/python3.6/site-packages/tqdm/_tqdm.py in __iter__(self)
   1020                 """), fp_write=getattr(self.fp, 'write', sys.stderr.write))
   1021 
-> 1022             for obj in iterable:
   1023                 yield obj
   1024                 # Update and possibly print the progressbar.

~/.conda/envs/poseIN/lib/python3.6/site-packages/torch/utils/data/dataloader.py in __next__(self)
    312         if self.num_workers == 0:  # same-process loading
    313             indices = next(self.sample_iter)  # may raise StopIteration
--> 314             batch = self.collate_fn([self.dataset[i] for i in indices])
    315             if self.pin_memory:
    316                 batch = pin_memory_batch(batch)

~/.conda/envs/poseIN/lib/python3.6/site-packages/torch/utils/data/dataloader.py in <listcomp>(.0)
    312         if self.num_workers == 0:  # same-process loading
    313             indices = next(self.sample_iter)  # may raise StopIteration
--> 314             batch = self.collate_fn([self.dataset[i] for i in indices])
    315             if self.pin_memory:
    316                 batch = pin_memory_batch(batch)

~/codedisk/dsl_py/pose-interpreter-networks/datasets.py in __getitem__(self, index)
     50 
     51         object_name = self.object_names_map[ann['category_id']]
---> 52         object_index = self.object_indices_map[object_name]
     53         object_id = self.object_ids_map[object_name]
     54 

KeyError: 'oilfilter'

How to read  annotations in datasets which only  used for pose estimation?

jimmyyhwu, 2019-04-13:
You can update the object list in the config file, like in [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/config/floating_kinect1_mask.yml#L6).

dingshenglan, 2019-04-13:
> 您可以在配置文件中更新对象列表，如此[处所示](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/config/floating_kinect1_mask.yml#L6)。

Really??I just think 

> You can update the object list in the config file, like in [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/config/floating_kinect1_mask.yml#L6).

I update the object list , but I met a new problem:

> RuntimeError: cuda runtime error (59) : device-side assert triggered at /opt/conda/conda-bld/pytorch_1535491974311/work/aten/src/THC/generated/../generic/THCTensorMathPointwise.cu:47

dingshenglan, 2019-04-13:
> You can update the object list in the config file, like in [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/config/floating_kinect1_mask.yml#L6).

I update the object list , but I met a new problem:

> RuntimeError: cuda runtime error (59) : device-side assert triggered at /opt/conda/conda-bld/pytorch_1535491974311/work/aten/src/THC/generated/../generic/THCTensorMathPointwise.cu:47

dingshenglan, 2019-04-13:
> You can update the object list in the config file, like in [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/config/floating_kinect1_mask.yml#L6).

I update the object list , but I met a new problem:

> --------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-11-fed55fe84c4c> in <module>
      3 with torch.no_grad():
      4     for input, target, object_index, object_id in tqdm(val_loader):
----> 5         position_error, orientation_error = forward_batch(model, input, target, object_index, object_id)
      6         position_errors.extend(position_error)
      7         orientation_errors.extend(orientation_error)

<ipython-input-10-8a93ac45140a> in forward_batch(model, input, target, object_index, object_id)
      5 
      6     position, orientation = model(input, object_index, object_id)
----> 7     position_error = (target[:, :3] - position).pow(2).sum(dim=1).sqrt()
      8     orientation_error = 180.0 / np.pi * pose_utils.batch_rotation_angle(target[:, 3:], orientation)
      9 

RuntimeError: cuda runtime error (59) : device-side assert triggered at /opt/conda/conda-bld/pytorch_1535491974311/work/aten/src/THC/generated/../generic/THCTensorMathPointwise.cu:47

jimmyyhwu, 2019-04-13:
Did you include the extra objects when training the pose estimation model? If it was trained on the original five objects, it won't work with a new object like oilfilter unless you include it in the training.

dingshenglan, 2019-04-13:
I just trianed on four objects, so how can I to use Oil change dataset to evaluate “end to end”？发自我的华为手机-------- 原始邮件 --------主题：Re: [jimmyyhwu/pose-interpreter-networks] Error about "end to end pose estimator" use "kinect2" in Oilchange datasets  (#13)发件人：Jimmy Wu 收件人：jimmyyhwu/pose-interpreter-networks 抄送：dingshenglan <15827291767@163.com>,Author Did you include the extra objects when training the pose estimation model? If it was trained on the original five objects, it won't work with a new object like oilfilter unless you include it in the training.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/jimmyyhwu/pose-interpreter-networks/issues/13#issuecomment-482853949",
"url": "https://github.com/jimmyyhwu/pose-interpreter-networks/issues/13#issuecomment-482853949",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

jimmyyhwu, 2019-04-13:
In that case, you'll want to modify the data loader to skip over objects you didn't train on.

Probably easiest is to modify [this line](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/datasets.py#L27) and pass in your four `object_id`s into `getAnnIds`.

```python
object_ids = [self.object_ids_map[object_name] for object_name in object_names]
self.object_instances = self.coco.loadAnns(self.coco.getAnnIds(imgIds=img_ids, catIds=object_ids))
```

You may need to reorder the code a bit to make it work.

Also see [here](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/data) for explanation of object ID vs object index.

dingshenglan, 2019-04-13:
Ok, Thanks for giving me this good suggestion. I'll make a try.发自我的华为手机-------- 原始邮件 --------主题：Re: [jimmyyhwu/pose-interpreter-networks] Error about "end to end pose estimator" use "kinect2" in Oilchange datasets  (#13)发件人：Jimmy Wu 收件人：jimmyyhwu/pose-interpreter-networks 抄送：dingshenglan <15827291767@163.com>,Author In that case, you'll want to modify the data loader to skip over objects you didn't train on.
Probably easiest is to modify this line and pass in your four object_ids into getAnnIds.
object_ids = [self.object_ids_map[object_name] for object_name in object_names]
self.object_instances = self.coco.loadAnns(self.coco.getAnnIds(imgIds=img_ids, catIds=object_ids))
You may need to reorder the code a bit to make it work.
Also see here for explanation of object ID vs object index.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/jimmyyhwu/pose-interpreter-networks/issues/13#issuecomment-482857613",
"url": "https://github.com/jimmyyhwu/pose-interpreter-networks/issues/13#issuecomment-482857613",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-04-14 by dingshenglan