jimmyyhwu/pose-interpreter-networks #16: Error when training pose_estimation about load pcd file.

yikakabu, 2019-04-27:
Hi, Jimmy. 
I have made my own dataset that contains 4 objects and pcd files. When I run:

> python train.py config/kinect1_mask.yml

There is an error occured:

> Traceback (most recent call last):
  File "train.py", line 295, in <module>
    main(cfg)
  File "train.py", line 215, in main
    criterion = PointsL1Loss(numpy_pcs).cuda()
  File "train.py", line 63, in __init__
    self._pcs = torch.from_numpy(np.array(numpy_pcs)).cuda()
ValueError: could not broadcast input array from shape (4,38720) into shape (4)

There are 38720 points in my first pcd file, and the format is same with your pcd files. It's like:

> VERSION 0.7
FIELDS x y z
SIZE 4 4 4
TYPE F F F
COUNT 1 1 1
WIDTH 38720
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 38720
DATA ascii
0 0.00193 0
0 0 0
0.0019499999 0 0
0.0038999999 0 0
0.0058499998 0 0
0.0077999998 0 0
0.0097500002 0 0
...

My config file is like :

> data:
        root: data/kinect1_mask
        pcd_root: objects
        num_subsets: 10
        val_subset_num: 101
        objects: [
            box,
            bottle,
            cola_can,
            power_bank
        ]
        batch_size: 32
        workers: 4
arch:
        num_input_channels: 1
        num_shared_fc_layers: 1
        num_shared_fc_nodes: 256
        num_position_fc_layers: 1
        num_position_fc_nodes: 256
        num_orientation_fc_layers: 1
        num_orientation_fc_nodes: 256
loss: points # l1 | posecnn | points_simple | points
optimizer:
        lr: 0.01
        lr_decay_epochs: [700, 1400]
        momentum: 0.9
        weight_decay: 0.0001
training:
        logs_dir: logs/
        checkpoints_dir: checkpoints/
        experiment_name: kinect1_mask
        print_freq: 10
        checkpoint_epochs: 100
        epochs: 2100
        log_dir:
        resume:

Could you give me any suggestions?
Thanks in advance.

jimmyyhwu, 2019-04-28:
The number of points must match in all of the pcd files. See [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/generate_pcds.ipynb) for my pcd preprocessing.

yikakabu, 2019-04-29:
That's the problem!
THANK you very much.

closed 2019-05-07 by yikakabu