jimmyyhwu/pose-interpreter-networks #18: Problem about pose_estimation training with my own dataset

LaureAcin, 2019-06-19:
Hi, Jimmy.

I use my own dataset which contains one object to train the pose estimation. When I ran train.py there is the error : 
RuntimeError: invalid argument 2: size '[-1 x 40960]' is invalid for input with 2768896 elements at /opt/conda/conda-bld/pytorch_1535491974311/work/aten/src/TH/THStorage.cpp:80

I went in the debug mode and I found that the problem is that the matrix x in models.py file at line 52 don't have the right size. Normally I should have matrix of shape [32,512,8,10] but I have [32,512,13,13] when I look x.shape. I don't understand how this matrix is created and so where does the problem come from. 

Could you give me any suggestion?

Thanks in advance.

jimmyyhwu, 2019-06-19:
I believe my input images were 640x480. If you would like to use a different image size, you can modify [this line](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/pose_estimation/models.py#L19). In your case, you could use `512 * 13 * 13`.

LaureAcin, 2019-06-26:
Thank you, that was the problem. Now I have an other problem. When I try to evaluate my trained model there is a problem with size again. In eval.ipynb at the line model.load_state_dict(checkpoint['state_dict']) I have this error : 
RuntimeError: Error(s) in loading state_dict for DataParallel:
	size mismatch for module.fc1.weight: copying a param of torch.Size([256, 40960]) from checkpoint, where the shape is torch.Size([256, 86528]) in current model.
	size mismatch for module.fc_p1.weight: copying a param of torch.Size([15, 256]) from checkpoint, where the shape is torch.Size([3, 256]) in current model.
	size mismatch for module.fc_p1.bias: copying a param of torch.Size([15]) from checkpoint, where the shape is torch.Size([3]) in current model.
	size mismatch for module.fc_o1.weight: copying a param of torch.Size([20, 256]) from checkpoint, where the shape is torch.Size([4, 256]) in current model.
	size mismatch for module.fc_o1.bias: copying a param of torch.Size([20]) from checkpoint, where the shape is torch.Size([4]) in current model.

I don't understand how I could change these sizes. 

Thanks in advance.

jimmyyhwu, 2019-10-02:
Closing due to inactivity.

closed 2019-10-02 by jimmyyhwu