jimmyyhwu/pose-interpreter-networks #20: ROS Dependencies

meghalD, 2019-10-01:
in order to run end_to_end_visualize.ipynb  I installed all the packages from conda and tried to run the ipynb file. Earlier it gave me pypcd error. I cloned the github repo of the same. But again got stuck somewhere and got an error 
RuntimeError: invalid hash value (expected "4bd2f8ea", got "f22d7c4219e69774cf0e6a9ebcaeb1a200420ae084036bf122712107be6576bb")

Please let me know how to resolve these ROS dependency and make the code run just to see how it works.

jimmyyhwu, 2019-10-02:
Sounds like a corrupted model file, you can try removing `~/.torch` and see if that helps. If that doesn't work, can you provide more details, such as the line where the error occurs?

By the way, there shouldn't be any ROS dependencies when running `end_to_end_visualize.ipynb`. You'll want to use an entirely separate conda environment for that, since the ROS package uses Python 2.7 instead of 3.6.

meghalD, 2019-10-03:
> Sounds like a corrupted model file, you can try removing `~/.torch` and see if that helps. If that doesn't work, can you provide more details, such as the line where the error occurs?
>
> By the way, [...]

for imports in the beginning I get an error 
from pose_estimation import utils as pose_utils
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
      from pypcd import pypcd
ModuleNotFoundError: No module named 'pypcd'

jimmyyhwu, 2019-10-03:
Check the README [here](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/pose_estimation) for installation instructions.

closed 2019-10-03 by meghalD