jimmyyhwu/pose-interpreter-networks #21: Issue with Segmentation Eval 403 Forbidden

SpencerTrihus, 2019-10-02:
Hi Jimmy,

I have downloaded the pretrained datasets, however when I attempt to run the segmentation eval Jupyter notebook. I receive an error in the third block. It seems that the code is downloading a checkpoint from a princeton website. Attempting to visit this site gives me a 403 Forbidden error. The address is: https://tigress-web.princeton.edu/~fy/drn/models/drn_d_22-4bd2f8ea.pth. 

Do users need special permission for this?

Thanks

jimmyyhwu, 2019-10-03:
I just updated the URL, give it another try!

SpencerTrihus, 2019-10-04:
I am still receiving the same error with the following line of code:
segm_model = segm_models.DRNSeg(segm_cfg.arch, segm_cfg.data.classes, None, pretrained=True)

I still receive a 403 Forbidden error. I have also attempted to simply open the link I posted above to the same result.

Thanks

jimmyyhwu, 2019-10-05:
Sorry for the confusion, you have to pull my new commit (51242d0) to update the URL.

closed 2019-10-15 by jimmyyhwu