jimmyyhwu/pose-interpreter-networks #22: end_to_end_visualize.ipynb HTTP 403 forbidden error

meghalD, 2019-10-03:
I loaded all the pretrained models and dataset required. While running end_to_end_visualize.ipynb I get an error at line 
segm_model = segm_models.DRNSeg(segm_cfg.arch, segm_cfg.data.classes, None, pretrained=True)

Downloading: "https://tigress-web.princeton.edu/~fy/drn/models/drn_d_22-4bd2f8ea.pth" 
 line 65, in load_url
    _download_url_to_file(url, cached_file, hash_prefix, progress=progress)
in _download_url_to_file
    u = urlopen(url)
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 532, in open
    response = meth(req, response)
  File "/usr/lib/python3.6/urllib/request.py", line 642, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.6/urllib/request.py", line 570, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 650, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 403: Forbidden

jimmyyhwu, 2019-10-03:
See @21.

jimmyyhwu, 2019-10-03:
See #21.

meghalD, 2019-10-03:
> See #21.

I don't know where is the change reflected. 
In the following line drn-d-22 is still forbidden. do I need some special permission?

model_urls = {
    'drn-c-26': webroot + 'drn_c_26-ddedf421.pth',
    'drn-c-42': webroot + 'drn_c_42-9d336e8c.pth',
    'drn-c-58': webroot + 'drn_c_58-0a53a92c.pth',
    'drn-d-22': webroot + 'drn_d_22-4bd2f8ea.pth',
    'drn-d-38': webroot + 'drn_d_38-eebb45f0.pth',
    'drn-d-54': webroot + 'drn_d_54-0e0534ff.pth',
    'drn-d-105': webroot + 'drn_d_105-12b40979.pth'
}

jimmyyhwu, 2019-10-03:
Pull the latest commit, 51242d0eb6f65630ec32934477f2225005fcf32e.

closed 2019-10-15 by jimmyyhwu