jimmyyhwu/pose-interpreter-networks #24: No such file or directory: 'data/OilChangeDataset\\annotations\\val_20171103_OilChange.json'

capslocknanda, 2019-10-04:
I can not unzip the oil change dataset. Its showing corrupted file. 
Kindly help, please !!!!

jimmyyhwu, 2019-10-05:
The download [script](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/download_data.sh) should automatically extract the files for you. I just checked and the script is still working.

You can run it using:

```bash
./download_data.sh
```

If the file is corrupted, you could try removing the downloaded files and running the script again.

capslocknanda, 2019-10-05:
It worked. Thank you very much.

closed 2019-10-06 by jimmyyhwu

arturohernandez10, 2022-01-09:
Hello,

I have downloaded the tar file several times. And I get the meshes and images folders. I've open files in both and they look good. But there are no annotations files. And I get the same error. Any suggestions?

jimmyyhwu, 2022-01-09:
They should still be in the tar file, but here is a zip with just the annotations: [annotations.zip](https://github.com/jimmyyhwu/pose-interpreter-networks/files/7836179/annotations.zip)