jimmyyhwu/pose-interpreter-networks #28: steps to train model again with another DRN

meghalD, 2019-10-31:
Hello, 
I want to train another drn model on the same dataset. Please guide me through steps for end to end evaluation.

jimmyyhwu, 2019-10-31:
I believe the segmentation [README](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/segmentation) should guide you through the steps.

meghalD, 2019-10-31:
I tried running python train.py config/drn_d_22_OilChange.yml. I throws an error as follows:

log_dir: logs/2019-10-31_03-23-27.170915_drn_d_22_OilChange
checkpoint_dir: checkpoints/2019-10-31_03-23-27.170915_drn_d_22_OilChange
loading annotations into memory...
Done (t=0.46s)
creating index...
index created!
loading annotations into memory...
Done (t=0.07s)
creating index...
index created!
Traceback (most recent call last):
  File "train.py", line 238, in <module>
    main(cfg)
  File "train.py", line 202, in main
    train_batch_time, train_data_time, train_loss = train(train_loader, model, criterion, optimizer, epoch)
  File "train.py", line 45, in train
    for i, (input, target) in enumerate(train_loader):
  File "/home/meghal/python_iros/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 336, in __next__
    return self._process_next_batch(batch)
  File "/home/meghal/python_iros/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 357, in _process_next_batch
    raise batch.exc_type(batch.exc_msg)
AttributeError: Traceback (most recent call last):
  File "/home/meghal/python_iros/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 106, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/meghal/python_iros/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 106, in <listcomp>
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/home/meghal/pose-interpreter-networks/segmentation/datasets.py", line 42, in __getitem__
    image, label = self.transforms([image, label])
  File "/home/meghal/pose-interpreter-networks/segmentation/transforms.py", line 211, in __call__
    return [self._transform(self.transforms[i], d, i) for i, d in enumerate(data)]
  File "/home/meghal/pose-interpreter-networks/segmentation/transforms.py", line 211, in <listcomp>
    return [self._transform(self.transforms[i], d, i) for i, d in enumerate(data)]
  File "/home/meghal/pose-interpreter-networks/segmentation/transforms.py", line 207, in _transform
    data = transform(data)
  File "/home/meghal/pose-interpreter-networks/segmentation/transforms.py", line 173, in __call__
    return adjust_gamma(data, gamma=value)
  File "/home/meghal/python_iros/lib/python3.6/site-packages/skimage/exposure/exposure.py", line 419, in adjust_gamma
    return out.astype(dtype)
AttributeError: 'Tensor' object has no attribute 'astype'

meghalD, 2019-10-31:
Googling for the error shows similar problems with other attributes. As far we have checked, none of them serve our purpose. Any pointers will be highly appreciated.

meghalD, 2019-11-01:
> Try downgrading `scikit-image`, I think some of the libraries have changed in the past year, breaking the code.
> 
> ```shell
> pip install scikit-image==0.14.2
> ```

It worked after downgrading to 0.14.2. Thanks.

closed 2019-11-01 by meghalD