jimmyyhwu/pose-interpreter-networks #30: Why do we need blender for demo?

vikas-baghel, 2020-06-15:
Hi, thanks for your great work. I wanted to see the results of your work, i.e. I will give a RGB image and expect 6dposes. I believe that blender was used to create the synthetic dataset for training Pose interpreter network. My question is why do we need blender for seeing the results.

My other question is if we require blender then how to use it on some online platform like google collab?

jimmyyhwu, 2020-06-15:
Blender is used to render visualizations of the pose predictions. If you don't need to visualize the predicted poses, then you could skip Blender.

I believe Blender should work in Colab, but I have not tried it and am not able to provide support for that.

vikas-baghel, 2020-06-15:
Thanks, I will try to find a way to use blender on colab.

closed 2020-06-15 by vikas-baghel