jimmyyhwu/pose-interpreter-networks #31: How can I get ground truth pose and the segmentation mask?

JZhao12, 2020-11-04:
Hi Jimmy

I have already downloaded the oil change dataset, I would like to ask how to get the ground truth poses and the segmentation masks from the .json file? (specifically, I want to get the ground truth pose and the segmentation mask of the blue oil funnel)

Thank you so much!

jimmyyhwu, 2020-11-05:
The json file follows the COCO dataset format. These two dataloaders might be useful:

Poses: https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/datasets.py#L21
Mask: https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/segmentation/datasets.py#L10

JZhao12, 2020-11-05:
Jimmy, thank you for your fast reply. Actually, I am working on a project related to 6d pose estimation, but I don't know how to generate your blue oil funnel images with your annotation. I can show you an example to explain what I mean. 
From Linemod dataset, this the rgb,
[image]
this is the mask
[image]
together with the .npy pose file(I cannot upload it due to format). Then I can train my network with them.

Now what confused me is, how can I export the segmentation mask and poses of the blue oil funnel, and save them as a file?  Just like the mask image I show you.

I think it's a quite prolixity work, I would appreciate it if you could help me.

Thank you so much Jimmy.

jimmyyhwu, 2020-11-05:
You'll have to write some code to export those, but it should be pretty straightforward. Most of what you need is in those two files I mentioned above.

Each `ann` in the dataset is an instance annotation for one object, and contains a mask and a pose.

For exporting the instance segmentation masks, see [this line](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/segmentation/datasets.py#L27). You can add a line to save `mask` to an image file. Something like:

```
if ann['category_id'] == 6:  # blue funnel
    Image.fromarray(mask).save(...)
```

Using `category_id` 6 will filter by blue funnel only, see [here](https://github.com/jimmyyhwu/pose-interpreter-networks/tree/master/data/README.md).

For the pose, it is simply `ann['pose']`, see [here](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/datasets.py#L42) for further processing that needs to be done. You can add a few lines right after your mask save code, and save those to a file too.

You also need to somehow iterate through the dataset to get all the `ann` instances. Probably the easiest way to do that is to modify [this notebook](https://github.com/jimmyyhwu/pose-interpreter-networks/blob/master/end_to_end_eval.ipynb).

For more about what fields such as `'image_id'` mean, see the COCO API [documentation](https://cocodataset.org/#format-data), particularly the "Object Detection" section.

Let me know if this works for you.

JZhao12, 2020-11-05:
Thanks Jimmy! It's really kind of you!

I'll try it immediately and update it to you if I get any progress on this.

Thank you so much!

closed 2021-06-24 by jimmyyhwu