jimmyyhwu/pose-interpreter-networks #33: Problem in training

gfdeng, 2023-06-04:


gfdeng, 2023-06-04:
Hello, sorry to bother after so long, but using the dataset you provided, there was a problem training the segmentation network. The downloaded dataset is complete, the image in the OilChangeDataset/image has 9829 images. 
It runs perfectly when running visualize.ipynb and eval.ipynb. But there is a problem in the running train.py, I wonder if you know what the problem is?

Traceback (most recent call last):
File "train_copy.py", line 238, in <module> main(cfg)
File "train_copy.py", line 202, in main train_batch_time, train_data_time, train_loss = train(train_loader, model, criterion, optimizer, epoch)
File "train_copy.py", line 45, in trainfor i, (input, target) in enumerate(train_loader):

torch\utils\data\dataloader.py", line 530, in __next__
data = self._next_data()
torch\utils\data\dataloader.py", line 1224, in _next_data
return self._process_data(data)
torch\utils\data\dataloader.py", line 1250, in _process_data
data.reraise()
torch\_utils.py", line 457, in reraise
raise exception
TypeError: Caught TypeError in DataLoader worker process 0.

jimmyyhwu, 2023-06-07:
If you haven't already, I would try running the code with the package versions from August 2018 (e.g., pytorch 0.4, torchvision 0.2).

closed 2023-06-07 by gfdeng